OpenScan-org/OpenScan2 #1: DOWNLOAD does not work

mcbriers, 2022-02-08:
Hi

The download button in Files & Cloud is broken.  It seems to populate the URL from the list, and not just the filename.  It gives a horrible reply: 

Cannot GET /%7B%22Set%22:%222022-02-07_20.36.08-89745.zip%22,%22Photos%22:100,%22Size%22:%22217.5MB%22,%22Date%22:%222022-02-07_20.36%22,%22Name%22:%2289745%22,%22Status%22:%22prepared%22,%22Size_full%22:217589003,%22id%22:4%7D

Cheers

MC

OpenScanEu, 2022-02-09:
Thanks! I just uploaded a new beta version, that you should be able to install via the Update&info tab in the interface. Just select 'beta' and 'check for updates' and confirm
(Your issue was caused by me adding an unnecessary link in the editor, which I removed through the update. But I fixed some other minor issues too)

could you confirm that the update works? Thank you!

mcbriers, 2022-02-09:
OK, thank you!  Been out for the evening, but just managed to perform the upgrade as described, and now the down;load works as expected.

I have not been able to test anything else yet, so if you broke the chain while fixing the wheel I won't know. ;)

OpenScanEu, 2022-02-10:
_broke the chain while fixing the wheel_ 
Never heard this one before. This might just be the perfect subtitle for my recent years work. At least, it is keeping me busy :))

mcbriers, 2022-02-10:
Hi Thomas

I can confirm that the entire workflow worked perfectly now.
Komischerweise the mini's rotor works correctly when scanning, but by
button controls it is inverted.  I may fix that for me, but is this a
'known' issue?

Also, the slider controls (Crop & number of photos specifically) are
extremely jumpy.  I find it impossible to get a consistent movement.  Is it
my mouse, or is NodeRed verarsching me?

Feel free to use the subtitle, I sucked it out of my thumb last night. ;)

Cheers

MC

OpenScanEu, 2022-02-23:
Which browser and device do you use to enter the user interface?

Concerning the motors: you can invert the direction in the settings menu under motor settings (which needs to be activated by switching to 'advanced settings' in the settings tab)

mcbriers, 2022-02-23:
I use Chrome, on Win10 64bit.

Changing the motor direction ins settings turns the scan sequence upside down, i.e. the rotor then starts at the top, and moves down.  Not a big issue, everything 'works', I am used to clicking the correct button for my system, already.

OpenScanEu, 2022-02-25:
The recent beta update fixed this issue. The message populating the download button contained some wrongly formatted information.

OpenScanEu, 2022-02-25:
@mcbriers concerning the motor direction, you might change the rotor_min and rotor_max angles to get the expected behavior. If not, feel free to open a new issue!
(I start to like github for that)

closed 2022-02-25 by OpenScanEu