OpenScan-org/OpenScan2 #2: Cloud registration error

sinnfeinn, 2022-02-18:
Hi Thomas, thanks for the great work. I have trouble to register for a cloud token:
I get:

**OpenScanCloud Registration - Failed
Registration failed, please try again.**

Any thing I can do on my part like logs to help?

thanks!

OpenScanEu, 2022-02-18:
I checked the logs and can see three successful requests (which i approved). How many times did you enter your data?
I will check the firmware for the reason of the shown error message

sinnfeinn, 2022-02-18:
Just received your email! Thanks! I tried it three times, once with a different email address.

OpenScanEu, 2022-02-25:
I have refined the server side and tried several times to reproduce the error, but it seems fixed :)

closed 2022-02-25 by OpenScanEu