OpenScan-org/OpenScan2 #4: MF button is on in AF mode

minimaker-nl, 2022-02-20:
The MF button is to the left (active) but there is no option for manual control. You only get this when the button is on the right. Should this be the AF button?

OpenScanEu, 2022-02-23:
The button to the left means manual focus is inactive (and autofocus is used instead). button to the right means MF active and the distance setting becomes visible + buttons enabled.
Btw. could you check the settable distance, i.e. placing a ruler in front of the camera, set the distance to 10cm and see if the focal plane is somewhere around this value.
You can see a larger version of the preview image at http://openscan/tmp/preview.jpg

OpenScanEu, 2022-02-25:
I am going to add more details to the information popups, which should make the various settings more clear.

closed 2022-03-03 by OpenScanEu