OpenScan-org/OpenScan2 #6: Arducam 16mp on raspberry pi4 not working

aharbick, 2022-02-22:
I followed the instructions here: https://en.openscan.eu/post/beta-firmware-imx519-16mpx-autofocus-dslr-finally-1 and it didn't show an image on the Scan screen despite reporting "--READY--" 

As requested here's the output of trying to capture an image from the command-line:
```
pi@openscan:~/OpenScan/tmp $ libcamera-still -n -t 1000 -o test.jpg --autofocus
[0:05:16.705895461] [1012]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:05:16.819083126] [1014]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx519'
[0:05:16.819271807] [1014]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:05:16.969493427] [1014] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/video0
[0:05:16.969626053] [1014] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0901 but control is not exposed by device /dev/video0
[0:05:16.969704959] [1014] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x00980911 but control is not exposed by device /dev/video0
[0:05:16.969817938] [1014] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009e0903 but control is not exposed by device /dev/video0
[0:05:16.974238252] [1012]  INFO Camera camera.cpp:937 configuring streams: (0) 2328x1748-YUV420
[0:05:16.975390686] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y10  Score: 500 (best 500)
[0:05:16.975540645] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y12  Score: 500 (best 500)
[0:05:16.975659772] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y14  Score: 500 (best 500)
[0:05:16.975903932] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BA10 Score: 1500 (best 500)
[0:05:16.976030262] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GB10 Score: 1500 (best 500)
[0:05:16.976161815] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BG10 Score: 1500 (best 500)
[0:05:16.976325459] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG10 Score: 1500 (best 500)
[0:05:16.976410845] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BA81 Score: 2500 (best 500)
[0:05:16.976493677] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BA12 Score: 500 (best 500)
[0:05:16.976575452] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GB12 Score: 500 (best 500)
[0:05:16.976656524] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BG12 Score: 500 (best 500)
[0:05:16.976738226] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG12 Score: 500 (best 500)
[0:05:16.976820928] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGB3 Score: 500 (best 500)
[0:05:16.976904000] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BGR3 Score: 500 (best 500)
[0:05:16.976989479] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GB14 Score: 500 (best 500)
[0:05:16.977074699] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt BG14 Score: 500 (best 500)
[0:05:16.977159512] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG14 Score: 500 (best 500)
[0:05:16.977244102] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GR14 Score: 500 (best 500)
[0:05:16.977328193] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGB4 Score: 500 (best 500)
[0:05:16.977406820] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pBAA Score: 1000 (best 500)
[0:05:16.977486207] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pGAA Score: 1000 (best 500)
[0:05:16.977566187] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pRAA Score: 1000 (best 500)
[0:05:16.977644018] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pgAA Score: 1000 (best 500)
[0:05:16.977727997] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGGB Score: 2500 (best 500)
[0:05:16.977808477] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pBCC Score: 0 (best 0)
[0:05:16.977888271] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pGCC Score: 0 (best 0)
[0:05:16.977969139] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pRCC Score: 0 (best 0)
[0:05:16.978048637] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pgCC Score: 0 (best 0)
[0:05:16.978132617] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pBEE Score: 500 (best 0)
[0:05:16.978217226] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pGEE Score: 500 (best 0)
[0:05:16.978301076] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pREE Score: 500 (best 0)
[0:05:16.978384999] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pgEE Score: 500 (best 0)
[0:05:16.978468071] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GRBG Score: 2500 (best 0)
[0:05:16.978553125] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GBRG Score: 2500 (best 0)
[0:05:16.978637882] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGBO Score: 500 (best 0)
[0:05:16.978721843] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y10P Score: 500 (best 0)
[0:05:16.978805859] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y12P Score: 500 (best 0)
[0:05:16.978889801] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt Y14P Score: 500 (best 0)
[0:05:16.978971299] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGBP Score: 500 (best 0)
[0:05:16.979054834] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGBQ Score: 500 (best 0)
[0:05:16.979138554] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RGBR Score: 500 (best 0)
[0:05:16.979222275] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt YVYU Score: 500 (best 0)
[0:05:16.979306939] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt YUYV Score: 500 (best 0)
[0:05:16.979384511] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt GREY Score: 2500 (best 0)
[0:05:16.979691596] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt VYUY Score: 500 (best 0)
[0:05:16.979769705] [1014]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt UYVY Score: 500 (best 0)
[0:05:16.980242230] [1014]  INFO RPI raspberrypi.cpp:624 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected mode: 2328x1748-pgCC
[0:05:16.982413083] [1014] ERROR IPARPI raspberrypi.cpp:567 Unable to find sensor control 0x009e0903
[0:05:16.982541154] [1014] ERROR IPARPI raspberrypi.cpp:376 Sensor control validation failed.
[0:05:16.982612189] [1014] ERROR RPI raspberrypi.cpp:1353 IPA configuration failed!
[0:05:16.982760334] [1014] ERROR RPI raspberrypi.cpp:761 Failed to configure the IPA: -32
[0:05:17.012504103] [1014] ERROR V4L2 v4l2_videodevice.cpp:1158 /dev/video1[13:cap]: Unable to request 4 buffers: Invalid argument
[0:05:17.012591823] [1014] ERROR RPI raspberrypi.cpp:821 Failed to allocate buffers
```

aharbick, 2022-02-22:
Hey!  I think I found a (partial) solution... https://forums.raspberrypi.com/viewtopic.php?p=1958297#p1958297

When I changed my /boot/config.txt and rebooted the libcamera-still command produced this output:
```
pi@openscan:~/OpenScan/tmp $ libcamera-still -n -t 1000 -o test.jpg --autofocus
[0:02:36.399452447] [1028]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0
[0:02:36.509947404] [1030]  WARN CameraSensorProperties camera_sensor_properties.cpp:141 No static properties available for 'imx519'
[0:02:36.510198931] [1030]  WARN CameraSensorProperties camera_sensor_properties.cpp:143 Please consider updating the camera sensor properties database
[0:02:36.663892317] [1030] ERROR DelayedControls delayed_controls.cpp:87 Delay request for control id 0x009a090a but control is not exposed by device /dev/video0
[0:02:36.667794111] [1028]  INFO Camera camera.cpp:937 configuring streams: (0) 2328x1748-YUV420
[0:02:36.668551673] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt RG10 Score: 2519 (best 2519)
[0:02:36.668651880] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt RG10 Score: 2148.24 (best 2148.24)
[0:02:36.668708178] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG10 Score: 1500 (best 1500)
[0:02:36.668760068] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 3819.24 (best 1500)
[0:02:36.668812737] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG10 Score: 5819.24 (best 1500)
[0:02:36.668870591] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt pRAA Score: 2019 (best 1500)
[0:02:36.668921648] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt pRAA Score: 1648.24 (best 1500)
[0:02:36.668973335] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pRAA Score: 1000 (best 1000)
[0:02:36.669024447] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 3319.24 (best 1000)
[0:02:36.669074708] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRAA Score: 5319.24 (best 1000)
[0:02:36.669398774] [1030]  INFO RPI raspberrypi.cpp:624 Sensor: /base/soc/i2c0mux/i2c@1/imx519@1a - Selected mode: 2328x1748-pRAA
[0:02:36.720588278] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: AfTrigger = 1
[0:02:36.720773821] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: Saturation = 1.000000
[0:02:36.721096165] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: Contrast = 1.000000
[0:02:36.721201317] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: Brightness = 0.000000
[0:02:36.721349544] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: AwbMode = 0
[0:02:36.721798096] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: Sharpness = 1.000000
[0:02:36.721906099] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: ExposureValue = 0.000000
[0:02:36.722064845] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: AeExposureMode = 0
[0:02:36.722298797] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: FrameDurationLimits = [ 33333, 33333 ]
[0:02:36.722532508] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: AeMeteringMode = 0
[0:02:36.722648271] [1034]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: NoiseReductionMode = 3
[0:02:36.724896308] [1030]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:36.724960625] [1030]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:36.725003015] [1030]  INFO RPISTREAM rpi_stream.cpp:122 No buffers available for ISP Output0
[0:02:36.933534296] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (6 left)
[0:02:36.971829407] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (5 left)
[0:02:37.000081439] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (4 left)
[0:02:37.032113054] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (3 left)
[0:02:37.065376930] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (2 left)
[0:02:37.098766755] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (1 left)
[0:02:37.132297918] [1030]  INFO RPI raspberrypi.cpp:1731 Dropping frame at the request of the IPA (0 left)
[0:02:37.969560803] [1028]  INFO Camera camera.cpp:937 configuring streams: (0) 4656x3496-YUV420
[0:02:37.970102098] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt RG10 Score: 1500 (best 1500)
[0:02:37.970281566] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt RG10 Score: 5971.24 (best 1500)
[0:02:37.970383847] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt RG10 Score: 9652 (best 1500)
[0:02:37.970481072] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt RG10 Score: 11971.2 (best 1500)
[0:02:37.970578038] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt RG10 Score: 13971.2 (best 1500)
[0:02:37.970682208] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 4656x3496 fmt pRAA Score: 1000 (best 1000)
[0:02:37.970778063] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 3840x2160 fmt pRAA Score: 5471.24 (best 1000)
[0:02:37.970873362] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 2328x1748 fmt pRAA Score: 9152 (best 1000)
[0:02:37.970966199] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1920x1080 fmt pRAA Score: 11471.2 (best 1000)
[0:02:37.971055905] [1030]  INFO RPI raspberrypi.cpp:122 Mode: 1280x720 fmt pRAA Score: 13471.2 (best 1000)
[... cut ...]
[0:02:38.053367383] [1036]  INFO IPARPI raspberrypi.cpp:654 Request ctrl: NoiseReductionMode = 2
Still capture image received
```

And actually took a photo:
[image]

The main Scan screen still doesn't show a live preview
[image]

When I tried to run the scan it completed and even produced a zip with the 50 images...  However it seems like maybe it's too zoomed in or cropped or something.  Here's the zip that was produced: https://www.dropbox.com/s/95krwe6wqpa6rdo/2022-02-22_17.21.23-apple.zip?dl=0

aharbick, 2022-02-22:
Aha!  Sorry for any confusion, but I think my problem was that for some reason the initial settings defaulted to CropX/CropY of 99%...  I adjusted the settings and the live preview started showing:
[image]

And I was able to produce a full set of images: https://www.dropbox.com/s/7gto8di4kxgg8qh/2022-02-22_17.35.30-apple-crop.zip?dl=0

Perhaps this issue is resolved now...

bodensee48, 2022-02-22:
cool i will try this out tomorrow. thx for ur testing

OpenScanEu, 2022-02-23:
I stumbled upon this forum post too and it looks great! I will start working on the implementation tomorrow (which should enable the use of the stock pi cameras too!)

aharbick, 2022-02-23:
@bodensee48 if you ssh into your pi and try `libcamera-still` does it work?  I specifically recommend this command `libcamera-still -n -t 1000 -o ~/OpenScan/tmp/still.jpg --autofocus`. Then make sure that you can view this: http://openscan.local/tmp/still.jpg

If that works then next try to do a scan... It may not show the live view, but it might still work.  Even after I posted above I have seen some measure of unreliability in whether the live view shows or not.  I haven't figured out how to diagnose that, but regardless once I did the stuff in the original beta release post and then changed the /boot/config.txt as above I have been able to get successful scans.

bodensee48, 2022-02-23:
Thomas send me his config.txt file no its works :-)

aharbick, 2022-02-23:
It didn't really seem like just an issue of lag.  I'm not really familiar with node-red but I was poking around the editor try to learn a thing or two and I noticed this:

[image]

Which seemed to me like a syntax error...  I doctored it up as:
[image]

and punched the deploy button and now the image shows after a short lag.  But I can't say I know what I'm doing :)

OpenScanEu, 2022-02-23:
@aharbick "But I can't say I know what I'm doing :)" --> same here
but in the end, it somehow works ^^

Your correction seems right and I must have messed up the html code while copy-pasting stuff...

closed 2022-03-03 by OpenScanEu