OpenScan-org/OpenScan2 #7: Suggestion: Sliders - also allow numerical entry?

minimaker-nl, 2022-02-23:
Changing settings to a specific value can be hard using the sliders. A nice to have would be to be able to enter values instead of using the sliders.

OpenScanEu, 2022-02-23:
This is possible, which sliders do you mean in particular?

sokol07, 2022-03-08:
I agree, honestly saying - all of them. Motors setting and scan settings mainly but I believe that numerical input would be nice everywhere.
However, you can also set the precise value using arrows keys - when you click on a slider it doesn't show up but is is "selected" and clicking right and left arrow key you can change the value by 1 step (it's a little feature of NodeRed).

minimaker-nl, 2022-03-09:
Nothing to add to sokol07's reply. Mainly scan and motor settings, but as an option it would be good for all places with sliders.

mickmorrison, 2022-06-17:
This would be a useful addition, particularly in the scanning window — using the scanner from a tablet with touch screen the sliders are not easy to control. Arrows work provided you can select the setting you wish to change. It works its just prone to creating mistakes so using ‘tab’ to nav through fields allowing manual input would go a long way towards improving usability.

closed 2024-11-11 by OpenScanEu