OpenScan-org/OpenScan2 #9: Downscale images for faster/reliable preview

OpenScanEu, 2022-03-01:
When connected by Wifi, the preview only works inconsistently --> it should help to create an additional setting, where you can set a downscale factor, so that the filesize is not as big.
(currently the preview window is trying to show full-res images and a 7Mb photo every 2s might be too much for most wireless connections)

sokol07, 2022-03-21:
Can this be achieved with modifying the "take_photo(file)" function in OpenScan.py? We could add function "take_preview_photo(file)", which would be an almost exact copy of the take_photo but with low quality parameter. In such a way we could save one file operation as we wouldn't take photo and convert it but rather just take a more lightweight photo.
I haven't tested it yet, it's just an idea, I'll try to test it when I have a free moment.

OpenScanEu, 2022-04-20:
@sokol07 after a long long time of being offline, I finally managed to get back to coding :)
I implemented this feature in the most recent update, where the image is downscaled (done here /home/pi/OpenScan/files/fla.py --> crop(), where the image is already opened for cropping and rotation. The extra step of downscaling does not take notably long, since the image file is already opened for processing.

closed 2022-04-20 by OpenScanEu