OpenScan-org/OpenScan2 #16: Impossible to light up all leds

sokol07, 2022-03-08:
Hi,
In the newest software version of OpenScan (beta for classic with arducam) it is impossible to turn on both LEDs at the same time.
Is it a matter of heat dissipation or power consumption?
If not, it can be easily solved in nodered by modifying the ringlight slider:
[image]
and consecutive code piece:
[image]

OpenScanFan, 2022-03-15:
or you could just change in line 12 to  "ringlight(2,True)" and not have to enter lines 13,14 and 15 like you have

sokol07, 2022-03-15:
In this case you would only have horizontal+vertical LEDs or all turned on, without option to have only diagonal ones. Adding the fourth case let's you have all possible combinations.

OpenScanFan, 2022-03-15:
yeah, I know, but why would you want horizontal or vertical anyway? what difference would it make?

sokol07, 2022-03-15:
I guess that may be useful - due to the object shape, maybe some polarizing benefits or something like that. I believe this may be useful at some point, maybe in future revisions different types of LEDs could be used, for example with different light color. That's how it's done in industrial vision systems, there are multiple diodes, for example red, blue, green and IR. I think this feature should be left as this could be useful at some point. For now: I think it may happen that choosing one set over another can give better result due to, for example, shadows.

closed 2022-04-20 by OpenScanEu