OpenScan-org/OpenScan2 #17: Cannot delete selected set

sokol07, 2022-03-08:
Hi,
On beta version of the OpenScan2, used with Rpi3 and Arducam, I am not able to delete selected one set of scanned files - I get the dialog, I select "yes" but nothing happens. The problem is in the way the files are deleted - they are found by name but in fact the real name contains not only the set name but also the date. Furthermore - the date in the filename is to the precision of seconds and the date in the files table is fetched to the precision of minutes. This issue can be fixed with the below code.

The fix is to modify two pieces in flow (in red frames):
[image]

First change is the `msg` in `del_set` flow. It code should be changed to:
```
msg.Set=global.get('set').Name
msg.Date=global.get('set').Date
msg.payload = 'Are you sure to delete and ALL associated files: ' + msg.Set + '?'
return msg
```

Second is the `del` function of `del_set` flow to this code:
```
import os
import shutil

dir="/home/pi/OpenScan/scans/"

if msg['payload']=="No":
    return

for i in os.listdir(dir):
    if msg['Date'] in i and msg['Set'] in i:
        if os.path.isdir(dir + i):
            shutil.rmtree(dir + i)
        else:
            os.remove(dir + i)
return msg
```

OpenScanEu, 2022-04-20:
@sokol07 Sorry, I really do not know how to merge the two branches, since I did a lot of changes in the meantime. For this reason, I just pushed my current version of the code. Could you maybe verify if the changes are working? 
btw, thank you so much for your effort! From now on, I will be more responsive and active again and start/continue working on the other repositories too :)

sokol07, 2022-04-20:
@OpenScanEu I have some thing to do today and tomorrow but I think I'll try to verify everything as soon as possible, definitely by the end of the week, and I'll notify you if any issue managed to survive in the new version of the code. :)