OpenScan-org/OpenScan2 #19: Automatic operation of the ringlight

sokol07, 2022-03-10:
I think it would be nice to have an option for the ringlight to turn off automatically after the job is finished. What I mean is - if you leave the scanner working at night it won't keep the ringlight turned on forever but it will turn it off as soon as the job is finished (and turn it on as soon as a new job is started).
To implement this I have added a few new nodes:
[image]

<details>
  <summary>Here is the json of the above blocks</summary>

```
[
    {
        "id": "8ebd1dcb5db156ed",
        "type": "ui_text",
        "z": "1613373abaf77a2c",
        "group": "7aaf184330605300",
        "order": 2,
        "width": 6,
        "height": 1,
        "name": "",
        "label": "Current Status:",
        "format": "<font color= {{msg.color}} > {{msg.payload}} </font>",
        "layout": "row-spread",
        "className": "",
        "x": 320,
        "y": 160,
        "wires": []
    },
    {
        "id": "94a7aec739f9266b",
        "type": "function",
        "z": "1613373abaf77a2c",
        "name": "loadS",
        "func": "var file = 'status_internal_cam'\nlet fs = global.get('fs');\nvar filepath = '/home/pi/OpenScan/settings/';\nconst data = fs.readFileSync(filepath+file, 'utf8');\nmsg.payload = String(data);\n\nif (data === 'no camera found'){\n    msg.color = 'red'\n}\n\nreturn msg\n\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 170,
        "y": 160,
        "wires": [
            [
                "8ebd1dcb5db156ed",
                "b87aecd9dc7f94b1"
            ]
        ]
    },
    {
        "id": "2415272f42ce468c",
        "type": "link in",
        "z": "1613373abaf77a2c",
        "name": "start status",
        "links": [
            "6c6ef2255a7d39e5"
        ],
        "x": 55,
        "y": 160,
        "wires": [
            [
                "94a7aec739f9266b"
            ]
        ]
    },
    {
        "id": "0ad3d8cfcb22472b",
        "type": "ui_switch",
        "z": "1613373abaf77a2c",
        "name": "",
        "label": "Ringlight test",
        "tooltip": "",
        "group": "7aaf184330605300",
[... 397 lines cut ...]
    {
        "id": "7aaf184330605300",
        "type": "ui_group",
        "name": "Settings",
        "tab": "e23b837a9f040895",
        "order": 1,
        "disp": false,
        "width": "6",
        "collapse": false,
        "className": ""
    },
    {
        "id": "e23b837a9f040895",
        "type": "ui_tab",
        "name": "Scan",
        "icon": "dashboard",
        "order": 2,
        "disabled": false,
        "hidden": false
    }
]
```
</details>

As the changes are a bit extensive I decided to fork OS to my github and I'll create a pull request soon, which will contain solution to this issue and also #16 and #17

sokol07, 2022-03-10:
There is an error in this solution, at the start of the flow there is no message for the LED function and everything crashes. I'm working on a fix.

sokol07, 2022-03-10:
I fixed the error, changed the code in the forked version and in the json code above.

OpenScanEu, 2022-04-20:
I have used a little different approach, where I added a idle-timer, which can be called through http://127.0.0.1:1312/ping which will return a time in seconds.
The timer will be reset to zero whenever the motor is being moved or the ringlight status changes. I set a timeout value to 300s, which can be modified by changing the file /home/pi/OpenScan/settings/timeout_ringlight
I intend to use the same timer (or a very similar approach) to turn off the stepper motors (which is currently not possible due to the hardware, but a new version should be out at some point)

sokol07, 2022-04-20:
How does it behave when the device is idling? If I leave the device for some time, return to it later and ai want to "wake up" the ring light I have to move the table or change the ring light switch position?
I have a feeling that's now intuitive, maybe it could be woken up by any action from UI?
My second concern is the timeout mechanism - if taking the picture takes longer time (because, for example, you're using focus stacking) you have to set a long timeout if i understand correctly. This way the ring light will also stay lit for long time after the job is done.
I don't say that's wrong approach and that it doesn't make sense, however I liked the way my solution worked, turning the ring light on/off as soon as the job was started or finished (I think the motors could be operated in future in the same way) - is there a reason you decided to switch to the timeout mechanism?

OpenScanEu, 2022-04-21:
> How does it behave when the device is idling? If I leave the device for some time, return to it later and ai want to "wake up" the ring light I have to move the table or change the ring light switch position?
The device won't wake up automatically, and you will have to use the ring light switch again

> My second concern is the timeout mechanism - if taking the picture takes longer time (because, for example, you're using focus stacking) you have to set a long timeout if i understand correctly. This way the ring light will also stay lit for long time after the job is done.
Yes, this is right. For this reason, I have set the timeout to a large value of 300s. So the ring light will stay on for 5 mins after the routine is done.

I have tried your solution, and it is much more efficient and elegant. But it somehow conflicted with my habit/typical workflow:

I turn on the device, turn on a ring light, spray the object and place it on the turntable, adjust the focus/brightness and start the routine. After a couple of minutes, I return to the device, change the object's orientation and just hit run to capture a second set. In the meantime, I prepare a second scan object.

So, long story short, I switch off the ring light only when powering off the device. So for me the timeout mechanism helps, that I do not forget the ring light (and in the future: motors), when I leave my place or forget to turn off the device. And I am too lazy to turn on the ring light again after each routine, when I need to adjust the brightness level for a new object.

sokol07, 2022-04-21:
Ok, I get it!

sokol07, 2022-04-21:
Ok, I get it!
Considering your workflow I understand that the timeout mechanism can be more useful.
However, maybe we could meet in the middle with the solution - maybe the timeout mechanism can be active only when status is "--READY--" so that setting the timeout to, for example, 1 second would turn the light off faster when the job is completed?

sokol07, 2022-05-17:
@OpenScanEu can we pull this value out from the config files to the settings page?

OpenScanEu, 2022-05-18:
@sokol07 Absolutely, I will do so in the update coming next week (only minor changes so far)

closed 2022-07-12 by sokol07