OpenScan-org/OpenScan2 #22: The device or resource (openscan) is not set up to accept connections on port "The World Wide Web service (HTTP)

OpenScanFan, 2022-03-14:
This has been driving up the wall and around the block, I have tried EVERYTHING to get this to work

Cant access the scanner via browser
It worked before, now just doesn't show up anymore...
Any idea what the issue is?

OpenScanFan, 2022-03-14:
Been using it for weeks now, it just suddenly stopped being detected from today
Using Ethernet
Using Beta+Arducam
Don't know how to ping
Have no idea how to check address

OpenScanFan, 2022-03-14:
Hey Sokol, thanks for trying to help, but like I said I have been using it everyday for weeks, no problems at all...just TODAY it went offline.
That documentation is old, the wpa_supplicant.conf file is no longer present in the boot directory

Used the Angry IP Scanner, cant seem to find the PI...hmmm...

[image]

sokol07, 2022-03-14:
The wpa_supplicant should be added and edited manually (it should have your WiFi credentials) as described in the docs. The template can be downloaded from the project google drive:
https://drive.google.com/drive/folders/1V_viMNELEG5r0Spcr9WuQKSPY0RPLPbV
Hard to tell why it's not reachable, maybe something caused some kind of a crash - do you have any signs of live from the scanner (lights flashing, etc)?
How are you connecting to it? Scanner to the internet router or directly to your computer?
If the RPi isn't reachable in any way maybe something wrong happened to the uSD card with the OS, you could try flashing the image to another card booting from it and trying to connect the scanner to check this.
Have you updated something befored it stopped working? Windows, Scanner OS or something?

sokol07, 2022-03-14:
No problem, I'm happy it worked! :)

closed 2022-04-20 by OpenScanEu