OpenScan-org/OpenScan2 #23: Turntable, turning in random directions

OpenScanFan, 2022-03-15:
Hey, hope someone can help with this one
When scanning, the turntable does not turn in equal distances, it just jumps all over the show
Any reason for this?

OpenScanFan, 2022-03-18:
Can someone please address this issue? its pretty serious for me @OpenScanEu or Tomasz? Anyone?

sokol07, 2022-03-21:
I briefly looked through the positions generator in code and I don't see an issue there. Are you sure that the software you're using are drawing this schematic precisely? Maybe it's just a matter of the software ability to detect the position?
Maybe I'll have some more time in the evening, I'll try to investigate the code generator further.

OpenScanFan, 2022-03-21:
Yes, I am sure the software I am using for reconstruction is calculating camera positions precisely. I can physically see the motor rotating in random positions on the scanner. 

Do you think it could be the voltage setting on the stepper motor driver on the board may be set incorrectly?

Yes please, if you can check it that would be great. thanks

sokol07, 2022-03-21:
I don't think so - even if the voltage (or rather - current) is set incorrectly a step is a step, I doubt that the stepper is missing steps unless you're scanning extremely heavy and it doesn't have enough power to move.
That's a stupid idea but I made that mistake and realised it after definitely too long time (please, don't laugh at me) - do you have a correct version of the OpenScan set in the setting? OpenScan Classic with Mini in setting is doing really strange things, I guess in the opposite way it is the same.
[image]

Further thing to check are the stepper parameters - at the default ones my steppers weren't working correctly. Check if the 90 degree rotation (manually by step) is really 90 degree - maybe there is something wrong with the steps/rev settting.

yoyo42, 2022-03-27:
I had noticed this but I assumed it was an algorithm that covered more angles in fewer images. Seems to work OK anyway.

closed 2022-04-20 by OpenScanEu

OpenScanEu, 2022-04-26:
@OpenScanFan 
[image]
Here you can see a comparison of the two approaches (turntable vs optimized).