OpenScan-org/OpenScan2 #24: Feature Request: Pause Scan

OpenScanFan, 2022-03-15:
It would be really handy to add a pause scan button
[image]

sokol07, 2022-03-21:
Ok, it took me a while but I managed to implement this.
In the SCAN flow a few new blocks are needed:
[image]

<details>
<summary>They need to have code as below: (click)</summary>

```
[
    {
        "id": "70e59042ca7c4455",
        "type": "ui_button",
        "z": "1613373abaf77a2c",
        "name": "pause",
        "group": "7aaf184330605300",
        "order": 27,
        "width": "4",
        "height": "1",
        "passthru": false,
        "label": "",
        "tooltip": "",
        "color": "",
        "bgcolor": "",
        "className": "",
        "icon": "fa-pause",
        "payload": "",
        "payloadType": "str",
        "topic": "topic",
        "topicType": "msg",
        "x": 330,
        "y": 1460,
        "wires": [
            [
                "2315704a2f3ba373"
            ]
        ]
    },
    {
        "id": "2315704a2f3ba373",
        "type": "file in",
        "z": "1613373abaf77a2c",
        "name": "read status",
        "filename": "/home/pi/OpenScan/settings/status_internal_cam",
        "format": "utf8",
        "chunk": false,
        "sendError": false,
        "encoding": "none",
        "allProps": false,
        "x": 470,
        "y": 1460,
        "wires": [
            [
                "7be623536d108880"
            ]
        ]
    },
    {
        "id": "7be623536d108880",
        "type": "switch",
        "z": "1613373abaf77a2c",
        "name": "",
        "property": "payload",
        "propertyType": "msg",
        "rules": [
            {
                "t": "eq",
                "v": "PAUSED",
                "vt": "str"
            },
            {
[... 204 lines cut ...]
system('mv '+ zippath + ' ' + basepath + 'scans/' + projectcode + '.zip')

save('status_internal_cam','Routine-done')

motorrun('rotor',position_last[0] - angle_start)
motorrun('tt',position_last[1])

save('status_internal_cam','--READY--')

msg['enabled'] = False
return msg
```
</details>

Finally, the dashboard elements needs to be placed in such configuration:
[image]

The pause button has got the pause icon and width of 4x1.

All this is implemented in my fork #20 

The result is a pause button:
[image]

It is active only when scanning is running, when device is paused the status will be changed to "PAUSED". The device can be resumed by clicking the pause button again or stopped by stop button.

OpenScanFan, 2022-03-24:
Sorry, but I have a couple of questions 

id": "70e59042ca7c4455",
        "type": "ui_button",
        "z": "1613373abaf77a2c",  **<--- z?**
        "name": "pause",
        "group": "7aaf184330605300", **<----what group is this? where do I enter it?**
        "order": 27, **<---where can I change the order?**
        "width": "4",
        "height": "1",
        "passthru": false,
        "label": "",
        "tooltip": "",
        "color": "",
        "bgcolor": "",
        "className": "",
        "icon": "fa-pause",
        "payload": "",
        "payloadType": "str",
        "topic": "topic",
        "topicType": "msg",
        "x": 330, **<----where do I enter these values?**
        "y": 1460,
        "wires": **<----where is this?**

Sorry, I am still learning this

sokol07, 2022-03-24:
You can import the blocks using the json code (https://nodered.org/docs/user-guide/editor/workspace/import-export)
The values you marked are not mean to be input by hand, I guess that group describes to which dashboard the button is added (my second screen, you can follow one of the tutorials about dashboards to find this window, for example: https://www.arubacloud.com/tutorial/how-to-create-a-node-red-dashboard.aspx), order describes position in the dashboard. X and Y are probably just positions of this block in the node red editor sheet and wires describe connections to other nodes in the flow (black lines connecting the blocks). I have no idea what is the z value but as I mentioned - I don't think you should bother, I believe that you can just import the blocks from the json provided, adjust the connectors and function code and place the button in right place in the dashboard layout list.

OpenScanEu, 2022-04-20:
@sokol07 @OpenScanFan I have added a pause button to the latest update.

closed 2022-04-20 by OpenScanEu