OpenScan-org/OpenScan2 #25: Feature Request: IMX230 support

OpenScanFan, 2022-03-16:
Anyone know how to install and use the IMX230 sensor with the scanner? Would be great to use it

xsiriox, 2022-03-23:
I just followed the guide to install OpenScan2 on raspbian, and it works on imx230. I just replaced the command related to the kernel driver installation with the one on the arducam site, for the imx230.

OpenScanFan, 2022-03-24:
Thanks, but sorry, I don't know anything about code. I know scanning though ;-)

Would you mind just laying out step by step (in detail) how you did it? it would really be helpful, not just for me, but for any other people out there like me who know nothing about this sort of thing

Thanks

xsiriox, 2022-03-24:
I think that very soon it will be fixed to be natively usable with imx230, but to get it working I did the following steps. 

- start from the latest pre-compiled image: https://drive.google.com/file/d/128VQ4KB6vNhLotbOsOKdBW8b1EHVzUw6/view?usp=sharing

- start web browser and open openscan interface. Go to setting and select the correct model of your hardware (opencscan mini or classic), and select imx519 camera. 

- click on reboot button and reboot the raspy.

- log in to your raspberry with ssh. You have to type the following 5 commands to install drivers :

wget -O install_pivariety_pkgs.sh https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh

chmod +x install_pivariety_pkgs.sh

./install_pivariety_pkgs.sh -p libcamera_dev

./install_pivariety_pkgs.sh -p libcamera_apps

./install_pivariety_pkgs.sh -p kernel_driver

- when it ask you to reboot, type y and then enter. 

All done, if you open openscan from web browser you should be able to do scans. 

In the scan page I just set the crop x and crop y to 40, but it's at your need. 

Let me know if it's working also for you. 

Bye

OpenScanFan, 2022-03-24:
Thank you so much, I will give it a try

Thanks again

OpenScanFan, 2022-03-27:
Did exactly what you said and turned out worse
The imx230 not available as an option and imx519 not being detected
[image]

xsiriox, 2022-03-27:
If you want email me, I will help you. blacksirius1985@yahoo.com

OpenScanEu, 2022-04-21:
Unfortunately, I can not test it, as I do not have said camera module.

I suspect the following solution, which is very similar to @xsiriox approach. It should not be necessary to install any drivers: 

(1) select the IMX519 camera in the settings menu --> This is necessary in order to use the focus controller 

(2) open the terminal and run ```sudo nano /boot/config.txt``` and modify the following line 

```dtoverlay=imx519,media-controller=1```

to

```dtoverlay=arducam,media-controller=1```

(maybe, you do not need meda-controller=1, but I would try it with it first)

(3) restart the device

It should be a very similar/same solution for the following modules: **IMX462, OV9281, OV2311, AR0234, OV7251, OG02B10, AR0234, IMX327, IMX298**

OpenScanFan, 2022-04-23:
OK, will give that a try, thanks

OpenScanFan, 2022-04-29:
Hey Thomas,

What software are you using to bring up the terminal?

OpenScanFan, 2022-05-01:
Gave it a try, no luck

I tried both ways, changed to dtoverlay=arducam,media-controller=1 and also without the media-controller=1

no differnece, just says camera not detected in OpenScan UI

xsiriox, 2022-05-01:
The only thing I can do is to send you a copy of my sdcard. For me is working.. Are using the arducam 21mpx?

OpenScanFan, 2022-05-01:
@xsiriox I finally got it to work, but the focus motor wont stop going back and forth and the live image feedback looks a little wavy, have no idea what's up with that....

xsiriox, 2022-05-01:
It's normal, because autofocus it's always on also for previews. Check issue #27. There is a fork of the project with a solution, or you can follow instructions to add the option to disable auto focus in preview. Find the right illumination too, it helps a lot for focus the object correctly.

closed 2024-11-11 by OpenScanEu