OpenScan-org/OpenScan2 #27: Auto Preview Update

xsiriox, 2022-03-23:
Hello! I'm using openscan mini with imx230, and I'd like to deactivate auto preview update. I don't want burn the autofocus motor, and the camera is doing a lot of noise because of autofocus is working on every preview shot. I'm using the imx519 setup, but with imx230 and autofocus is working fine. Please add the function or tell me what to modify in the code to disable the preview! 

Thank you

sokol07, 2022-03-24:
Are you familiar with the node red?
You should edit the SCAN flow, adding two blocks:
[image]
Code for the added function node:
ON START (this is important!)
```
global.set('enable_preview',true)
```
ON MESSAGE
```
global.set('preview_enable',msg.payload)
```

You should also modify the "enable" function below the marked nodes on the screenshot above:
```
msg.flag = global.get('flag')
if (global.get('flag_pw')== true && global.get('preview_enable')== true){
    return msg
}
```

The "preview enable" switch should be placed in placed in "[Scan] Settings" dashboard:
[image]

And then you should place it wherever you want in the dashboard, for example:
[image]

sokol07, 2022-03-24:
I've added this to my fork, included in #20

xsiriox, 2022-03-24:
Wow! Thankyou!!! 

So fast!!! Il try today!!

sokol07, 2022-03-24:
No problem, I hope it'll work as expected. :)
I've tested it only with Arducam, if there is any problem - let me know.

closed 2022-07-12 by sokol07