OpenScan-org/OpenScan2 #28: Feature Request: Timer

OpenScanFan, 2022-04-03:
I recently realized it would be really handy to have a time duration and time remaining for the scan
its because I need to plan my day(s) around scan times, I scan all day everyday

Also maybe a total time duration next to the saved file name?

OpenScanFan, 2022-04-20:
Thank you for your reply Thomas,
Yes, I use the scanner all day everyday, great scanner

ok, then would it be possible to put a time stamp on the finished file? i.e. the total time it took to do the scan?

OpenScanEu, 2022-04-21:
@OpenScanFan What do you think about the following:
[image]
Where the first number 314 is the remaining time in seconds. And 357s the total estimated time from start to finish. It updates after each photo, but needs roughly 10-20 photos to get a good precision (+- 10% after 10 photos, +- 3% after 20 photos).

would you still want the timestamp on the filename?

OpenScanEu, 2022-04-21:
I improved the ETA calculation and so it is quite accurate after 5 photos.

closed 2022-04-21 by OpenScanEu

OpenScanFan, 2022-04-23:
Hey Thomas,

Sorry for my late reply,

hmmm....giving the total time in seconds...that may be useful for someone else, but not for me

I need a shorter hand version ie. hours and minutes and then seconds

[image]

OpenScanEu, 2022-04-25:
@OpenScanFan how about: 
[image]
(But I won't add those numbers to the filename, as it is already long enough ;)

OpenScanFan, 2022-04-25:
That's good enough for me, thank you so much Thomas, it's a big help