OpenScan-org/OpenScan2 #29: Feature Request: IMX682 (64MP) supoport

OpenScanFan, 2022-04-21:
Was wondering if you could start implementing support for the new Arducam IMX682 Hawk-Eye 64Mp module?

It is mentioned that only the 16mp super-resolution will be supported with Raspberry Pi 3b+ and down, only the Raspberry Pi 4 can support the full 64Mp

OpenScanEu, 2022-04-21:
@OpenScanFan after quite some trouble, I finally got the camera to kinda work, BUT it has been quite a pain... And so far, the results do not seem significantly better compared to the 16mpx version.

I will do some more testing, before trying to implement it. The camera works most of the time, but after 50-150 photos it somehow freezes and does not return an image. I am still waiting for an answer/fix from Arducam, but so far no luck.

I will keep you posted if there is any progress.

OpenScanFan, 2022-04-23:
Ok, great, thanks.
Would be great if you could get it to work fine
Keep me posted

OpenScanFan, 2022-07-12:
@OpenScanEu Any news? Any progress made?

closed 2024-11-11 by OpenScanEu