OpenScan-org/OpenScan2 #32: Feature Request: Histogram

OpenScanFan, 2022-04-24:
A histogram would be very useful for more accurate scanning and getting more correct colours

OpenScanEu, 2022-04-25:
Hmmmm, on the one side I really like this idea, on the other side, I fear the changes needed to implement it properly.

From a quick search, it should be possible with the currently used PIL library: https://www.geeksforgeeks.org/python-pil-image-histogram/, but I have no idea, how and where a useful histogram could be displayed onto the user interface...

My main has been to simplify the user interface as much as possible and move most options to the settings menu or elsewhere (out of reach for most new users, as they get distracted so easily...).

OpenScanFan, 2022-04-25:
It's no rush

Just display it in the top right or left of the live preview image. You can add a show hid histogram button

Just an idea, but if you could implement this one day, it would be super useful

sokol07, 2022-05-17:
I don't know whether the histogram will be really usefull - the only part of the image that interest us is the object but the histogram will also contain information about the background - on the histogram it can "hide" the object information even though the object will be distinguishable. Moreover, I believe the setting of the exposure "by eye" is enough for the application, the most important thing is not to have too dark or too bright areas.
Maybe it would be more beneficial to have warning if there are too light or too dark parts in the image with option to preview the too light/dark areas? (if the too dark area is a black background we don't care anyway)
I think the easies way to achieve this is to convert the image to grayscale and mark all pixels which value is out of range (for example, higher than 245 or lower than 10).

closed 2024-11-11 by OpenScanEu