OpenScan-org/OpenScan2 #33: Feature Request: Focus Peaking

OpenScanFan, 2022-04-24:
Focus peaking would be very useful for more accurate focus stacking around the whole object

OpenScanEu, 2022-04-25:
Another interesting idea. To be honest, I have never heard of focus peaking before and this looks quite useful.
It can be done, but I am not really capable of doing it efficiently using the included PIL library:
https://stackoverflow.com/questions/60872630/get-in-focus-pixels-of-an-image

It took an additional 5-8s for one image:
[image]
(left: original, middle:detected edges, right:overlay)

So, I am not sure, whether it is worth the delay.

sokol07, 2022-05-17:
I guess the RPi processor is a little too small for the live image processing...
But maybe it would be possible to add a button for the focus peaking in preview? We could set up the whole scan, light, camera, etc. and before starting the scan we just hit the button and check where the focus is, maybe do some precise MF tuning if we're using it - in this case the delay is something we can tolerate. Than, during the execution of the scan the peaking can be turned off not to cause any delay.
It would be necessary to have two options - one would turn on/off the focus peaking and the second option would be the color of the peaking (you don't want a red overlay on a red object).

OpenScanEu, 2022-05-18:
@sokol07 I agree, that this could be an option, which should be activated via a switch.
The image processing is currently done by the local flask server /home/pi/OpenScan/fla.py under /crop.
So it would be relatively simple to add a few (optional) lines to further process the images (at the moment it is only downscaled and/or rotated).
I just did not manage to enhance the detected edges enough, so that it would be clearly visible in the overlay (and I did not manage to change the color of said overlay, yet)

sokol07, 2022-05-18:
I did some edge detection previously in python, I guess I could try to get back to these codes and implement them to see what I can achieve - if it will be anything useful, I'll share it here.

closed 2024-11-11 by OpenScanEu