OpenScan-org/OpenScan2 #44: needs autofocus targeting

jank0cisco, 2022-06-17:
the arducam is great because it has autofocus but it does sometimes focus on things in the distance. like the other side of the ring on my mini or something through the gap. is there a way to fix this as it mostly happens at the start of the scan.

OpenScanEu, 2022-06-17:
@jank0cisco There are a few things you can do:
(1) make sure that the object is properly centered. The provided turntable holders have a number on it, indicating the object size. E.g. you have an object of roughly 5 cm height --> use the turntable labeled 5
(2) for small objects <3cm it is better to set the focus manually. The autofocus procedure seems very sensitive at the center of the image. Small objects tend to be off-center in some of the photos (which could be reduced by (1))
(3) something I personally never used, but seen many people using: A unicolor backdrop or light tent. The reasoning is, that if the background has little to no feature, the camera has nothing to focus on.

Hope that helps :)

OpenScanEu, 2022-06-17:
@jank0cisco Just remembered this one: https://www.reddit.com/r/OpenScan/comments/ibujfa/third_scan_result_with_mini_plus_custom_mod/

jank0cisco, 2022-06-18:
thanks found the stl for the dome above thought id post a link to help people in the future
https://www.thingiverse.com/thing:4587217/files