OpenScan-org/OpenScan2 #46: Issue: Cant Connect to scanner

OpenScanFan, 2022-06-24:
Hey,

Just ran in to an issue with my OpenScan Mini

It had been working fine for ages...until one day, it stopped connecting. I have tried the following
1. Connecting via ethernet and typing in http://openscan/ - result is, "This site cannot be reached"
2. tried reformatting the SD card and burning the new image and the WPA Supplicant with all prior working data included - result Angry IP Scanner cannot detect the scanner over the Wi-Fi

What is going on? it was working just fine, then "poof!!" stopped working....this is a real mystery to me, please help

josepmy, 2022-06-24:
The wifi password is the same as when it worked?

OpenScanFan, 2022-06-25:
Everything is EXACTLY the same, nothing has changed

sokol07, 2022-07-12:
Have you tried connecting with Ethernet cable, directly from PC to RPi? If you connect directly over ethernet, can you find the scanner IP and ping it?

OpenScanFan, 2022-07-12:
It wouldn't detect it at all, can't ping it

sokol07, 2022-07-12:
This suggests there is some problem with RPi... What RPi are you using? Have you checked with another SD card? (maybe the SD card is faulty?)

OpenScanFan, 2022-07-14:
Changed RPi board and SD card all combinations, no results

Polectron, 2022-07-14:
Can you connect your RPi to a screen and check if its booting properly?

OpenScanFan, 2022-07-14:
I did not try that, will do
Although I did try 2 Pi's and nothing...

sokol07, 2022-08-02:
The issue seems like a problem with particular unit (PCB maybe?), not the project as a whole.
For faster support I suggest checking out project's Discord server (a lot of helpful people there!) - if the problem hasn't been solved yet I hope we will be able to help you there: https://discord.gg/muJJDQuv

closed 2022-08-02 by sokol07

Dufrenoy, 2024-12-29:
Same issue. I spent many time to use it. With the available image : not working. I install via from scratch. succeed to connect but after update, it was impossible de reach the web ui and to connect via SSH. Openscan is very disapointing.

ip4368, 2025-06-28:
I had the same issue just now, but realising it was because me using MTU 9000, tldr, rpi4 doesn't support MTU 9000, so if you have set your computer MTU to 9000 (or jumbo frame), you just need to set it back to 1500.