OpenScan-org/OpenScan2 #72: Manual Focus doesn't work

AndreasEhrhard, 2023-01-28:
Disabeling the AUtofocus-Funktion of the adru-Cam doesn't work on Mini and also on Classic.
I changed the cameras, the Cable, the Raspberry Piy, ... 
What can I do?

winne27, 2023-04-04:
The call to take a photo in the software uses wrong options. Soon I will provide a patcxh for that.

apurvakalia, 2023-04-07:
One problem I found was the wrong access to the options dictionary ctrl in the file Arducam.py. It tries to access a key called "default" whereas the correct key value is "default_value". You can make that change manually in the code (I believe it is line 153 in Arducam.py)

winne27, 2023-04-07:
Maybe this is the solution:

have a look to patch

https://github.com/OpenScan-org/OpenScan2/pull/75/files#diff-7572f6a8aafa0d7fce069e4ec04593fc585cebd3e72d421bd8eac40284ff5b35

or 

replace OpenScan.py by
 [update/mini/OpenScan.py](https://github.com/winne27/OpenScan2/blob/5082aac17edbef6a911509b5f9ad820d141dec78/update/mini/OpenScan.py)

apurvakalia, 2023-04-10:
I tried the OpenScan.py mentioned above, but still running into problems:
- when using manual focus, I can see the camera focused in preview, but it reverts back to defocused mode again. I tried using stack option to enable to SET button but that does not help
- autofocus does not work at all. My object size is medium, but the camera keeps focusing on more distant objects
Any ideas?

winne27, 2023-04-10:
Did you replace OpenScan.py in /usr/lib/python3/dist-packages/ ?

AndreasEhrhard, 2023-04-11:
Hi!
I can#t work at the moment: Holliday on TEXEL.
Will try it, when I#m backmuome.

Regards
Andreas

closed 2024-11-11 by OpenScanEu