OpenScan-org/OpenScan2 #74: Autofocus not working

winne27, 2023-04-04:
The timeout option (-t)  in calling libcamera-still in OpenScan.py raises non focusing for imx519. By debugging and testing two solutions were found:

1. allow in Settings timeout to be higher than 1. With 2 or 3 autofocus will work
2. remove option -t from line 158 in OpenScan.py

Both worked for me.

apurvakalia, 2023-04-10:
Settings does not allow timeout to go beyond 1. How did you set it to 2 or 3? BTW, I see that you have removed -t in OpenScan.py from the other thread

winne27, 2023-04-10:
This limit is set somethere in the nodered configuration. I don't where.

I was testing it on command line with

`sudo libcamera-still -n  --sharpness 0 -o /home/pi/OpenScan/tmp/tmp.jpg --shutter 116000 --saturation 1 --contrast 1 --awbgains 0,0 --gain 1 -q 95 --autofocus-mode auto -t 2000 `

The parameter -t is the timeout in milliseconds. I tested until /home/pi/OpenScan/tmp/tmp.jpg becomes sharp. Removing this option was also successful fpor me.

apurvakalia, 2023-04-10:
I will try it on the command line as well. Removing the option does not seem to help for me. BTW, look at issue #44 on this. Maybe that is the solution...

closed 2024-11-11 by OpenScanEu