OpenScan-org/OpenScan2 #78: Lost connection to my openscan mini

Triguard2018, 2023-05-30:
We assembled my openscan mini on the weekend. 

We connected it to our home network with an ethernet cable to our network switch and I ran a first scan of 100 pictures. It appeared to be working well.

Because of the location of the home network switch, the initial setup could only be temporary, so I moved the openscan mini into another room and tried to connect via wifi.
When I attempted to connect to the page - http://openscan/ - it could not be found and I got the result "Hmmm… can't reach this page".

I took the openscan mini back to its original position and reconnected it to the home network switch with the ethernet cable. I got the same result "Hmmm… can't reach this page".
I have tried to connect the device directly to my laptop with both an ethernet cable and USB cable, but still get the same result.

Right now I cannot connect to the device and do any scanning. It must be something I have done, but I can't figure out what.

Does anyone have any ideas about how to fix this problem?

Thanks

OpenScanEu, 2024-11-11:
in some networks, it is necessary to use http://openscan.local instead. or use the ip address of the device (see your router or use angryIPscanner to find the ip)

closed 2024-11-11 by OpenScanEu

Dufrenoy, 2024-12-29:
Openscan mini is a scrap a try to use it spent money ... nothing is working and price for diy kit is expensive. I'm quite disapointed. I do not recommand the Openscan mini